daniel-thompson/i2c-star #6: use cortex-m-generic.ld instead of libopencm3_stm32f4.ld

gdamjan, 2018-08-27:
this seems like an upstream change in libopencm3, but this helped me compile your project.

there's no libopencm3_stm32f4.ld anywhere

```
diff --git a/src/bootloader/lctech-f103.ld b/src/bootloader/lctech-f103.ld
index 56dde87..a6e8e9e 100644
--- a/src/bootloader/lctech-f103.ld
+++ b/src/bootloader/lctech-f103.ld
@@ -28,5 +28,4 @@ MEMORY
 }
 
 /* Include the common ld script. */
-INCLUDE libopencm3_stm32f1.ld
-
+INCLUDE cortex-m-generic.ld
diff --git a/src/lctech-f103.ld b/src/lctech-f103.ld
index fc5714e..00b739d 100644
--- a/src/lctech-f103.ld
+++ b/src/lctech-f103.ld
@@ -28,5 +28,4 @@ MEMORY
 }
 
 /* Include the common ld script. */
-INCLUDE libopencm3_stm32f1.ld
-
+INCLUDE cortex-m-generic.ld
diff --git a/src/stm32f4-discovery.ld b/src/stm32f4-discovery.ld
index 927c786..c72a9bf 100644
--- a/src/stm32f4-discovery.ld
+++ b/src/stm32f4-discovery.ld
@@ -28,5 +28,5 @@ MEMORY
 }
 
 /* Include the common ld script. */
-INCLUDE libopencm3_stm32f4.ld
+INCLUDE cortex-m-generic.ld
 
```

zoobab, 2018-09-03:
I am trying to compile the project with the following Dockerfile:

```
FROM ubuntu:18.04
LABEL Description="I2C-star for STM32 Bluepill board"          

RUN DEBIAN_FRONTEND=noninteractive apt-get update -y -q && apt-get install -y -q sudo make python gcc-arm-none-eabi git-core libnewlib-arm-none-eabi

ENV user i2cstar
RUN useradd -d /home/$user -m -s /bin/bash $user
RUN echo "$user ALL=(ALL) NOPASSWD:ALL" > /etc/sudoers.d/$user
RUN chmod 0440 /etc/sudoers.d/$user

USER $user
WORKDIR /home/$user
RUN mkdir -pv code
COPY . ./code/
RUN sudo chown $user.$user -R /home/$user/code
WORKDIR /home/$user/code/
RUN git submodule update --init --recursive
RUN make
```

Then I end up with a similar error:

```
  CC      sync.c
  CC      dwt.c
  AR      libopencm3_stm32f4.a
  BUILD   src/bootloader
  CC      i2c_ctx.c
  CC      bitops.c
  CC      fibre.c
  CC      fibre_default.c
  CC      list.c
  CC      messageq.c
  CC      regdump.c
  CC      ringbuf.c
  CC      time_libopencm3.c
  CC      util.c
  CC      usbdfu.c
  LD      usbdfu.elf
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: cannot open linker script file libopencm3_stm32f1.ld: No such file or directory
collect2: error: ld returned 1 exit status
../../mk/libopencm3.rules.mk:158: recipe for target 'usbdfu.elf' failed
make[1]: *** [usbdfu.elf] Error 1
Makefile:75: recipe for target 'src/bootloader' failed
make: *** [src/bootloader] Error 2
The command '/bin/sh -c make' returned a non-zero code: 2
```

Any idea how to fix it?

daniel-thompson, 2018-09-04:
@zoobab: I can't see a `git clone` in the Docker recipe: it the source being copied from somewhere outside of the container (and did you update the source to a newer version of libopencm3)?

I just did a fresh clone as a test and it builds fine for me. I think the changes from @gdamjan are needed with newer versions of libopencm3 (where they have cleaned up the linker files, but not the libopencm3 that is fetched by default).

@gdamjan: Thanks for the patch. I'll take a look as soon as I get the time!

daniel-thompson, 2018-09-04:
@gdamjan: Actually having diffed the .ld files in libopencm3 these changes are "obviously correct". Do you want to submit a PR or shall I just repeat the changes myself.

gdamjan, 2018-09-04:
@daniel-thompson 

I don't think I did anything special with the repo, just cloned master, and I have the submodules as added in your repo:

```$ git submodule status 
 ea6a5a7efd5e83b6a7e2b55506acbc863a6d0c34 libopencm3 (ea6a5a7e)
 0132ee224460b07c45598f55075e723188fc377d librfn (0132ee2)
```

gdamjan, 2018-09-04:
@daniel-thompson I'll send a PR

daniel-thompson, 2018-09-04:
> @daniel-thompson 
> 
> I don't think I did anything special with the repo, just cloned master, and I have the submodules as added in your repo:

Sorry, my mistake. Tab completed into the wrong directory...

closed 2018-09-04 by daniel-thompson

n01287334, 2020-11-25:
@zoobab Hey, I get the same issue, how did you fix it?

n01287334, 2020-11-26:
@gdamjan Hi, could you explain what file to edit?

gdamjan, 2020-11-26:
The PR that fixes this has been merged 2 years ago

PR #7 (2018-09-04): Use cortex-m-generic.ld instead of libopencm3_stm32f4.ld
fixes #6

this is an upstream change in libopencm3, there's no libopencm3_stm32f4.ld but cortex-m-generic.ld
commit 90091602 (2018-09-04): use cortex-m-generic.ld instead of libopencm3_stm32f4.ld

fixes #6

this is an upstream change in libopencm3, there's no libopencm3_stm32f4.ld but cortex-m-generic.ld