daniel-thompson/i2c-star #8: Dockerfile attempt does not build under Ubuntu 18:04

zoobab, 2018-09-04:
Hi,

Just tried to compile the head of the git with a Dockerfile:

https://gitlab.com/zoobab/i2c-star/blob/master/Dockerfile

I ran the CI at gitlab (sorry there is no CI afaik on github, so I mirrored it to Gitlab):

https://gitlab.com/zoobab/i2c-star/-/jobs/94547265

Which ends up with this error:

```
  CC      time_libopencm3.c
  CC      util.c
  CC      i2c-explorer.c
  LD      i2c-explorer.elf
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: i2c-explorer.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libc.a(lib_a-abort.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libc.a(lib_a-abort.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: i2c-explorer.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libc.a(lib_a-ctype_.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libc.a(lib_a-ctype_.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: i2c-explorer.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libc.a(lib_a-errno.o) does not
```

Any idea how to fix it?

daniel-thompson, 2018-09-04:
This looks like it is likely to be a problem with your toolset (it does not support -mfloat-abi=hard).

It would be best to adopt a better toolset but you can probably workaround by not building the STM32F4 applications:

    make src/bootloader src/i2c-stm32f1-usb

Alternatively you could hack at the FPU settings:

    make FP_FLAGS=-mfloat-abi=soft

zoobab, 2018-09-04:
With "make src/bootloader src/i2c-stm32f1-usb" the build works now.

daniel-thompson, 2019-01-28:
Hope this was solved but I'm going to close it on the assumption it was a docker issue (e.g. nothing to do with i2c-star).

closed 2019-01-28 by daniel-thompson