daniel-thompson/i2c-star #19: cant Make...

Zombi13, 2023-04-14:
when i try to do
/i2c-star$ make

I get

  GENHDR  include/libopencm3/stm32/f1/irq.json
/usr/bin/env: ‘python’: No such file or directory
make[1]: *** [Makefile:59: include/libopencm3/stm32/f1/irq.json.genhdr] Error 127
make: *** [Makefile:57: libopencm3] Error 2

when i try to call python
/i2c-star$ python
Command 'python' not found, did you mean:
  command 'python3' from deb python3
  command 'python' from deb python-is-python3

 System:
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.2 LTS
Release:        22.04
Codename:       jammy

how to fix it  ???

daniel-thompson, 2023-04-16:
Sharing a PR to update to the latest version of libopencm3 would probably solve it (I haven't checked but I assume later versions of libopencm3 use python3 to generate this stuff).

However you could probably also get this to build by making python2 the default python interpreter: https://linux.how2shout.com/how-to-install-python-2-on-ubuntu-22-04-lts-jammy-linux/

flaviut, 2023-11-01:
Here are some pre-built binaries for 3ad87f38fa834521caf606c0e0b4d87e16531231: [i2c-star-built.tar.gz](https://github.com/daniel-thompson/i2c-star/files/13223124/i2c-star-built.tar.gz)

Build seems to have worked fine for me, even though I don't have python2 installed.

daniel-thompson, 2023-11-10:
I think that means apt install python-is-python3 will probably fix this.

closed 2023-11-10 by daniel-thompson